DustinWatts/FreeTouchDeck #126: Problem with ledcSetup while uploading FreeTouchDeck.ino

OpossumTinte, 2024-06-19:
Hey, 

I get an error in this section while uploading the .ino data.

  ledcSetup(0, 5000, 8);
#ifdef TFT_BL
  ledcAttachPin(TFT_BL, 0);
#else
  ledcAttachPin(backlightPin, 0);
#endif // defined(TFT_BL)
  ledcWrite(0, ledBrightness); // Start @ initial Brightness

Error: 'ledcSetup' was not declared in this scope

TY in advance

DustinWatts, 2024-06-26:
The quick solution is to downgrade the ESP32 Arduino core. Because there are some breaking changes regarding ledc in version 3. If you go for something like 2. something, It will work again. The long time solution is something I am working on but it can take some time due to being really busy...

OpossumTinte, 2024-06-29:
Mate, you are a bloody legend. TY, I love my touchdeck!

DustinWatts, 2024-06-30:
You are welcome mate! :)

It is just a temporary fix, but it does work. I'm keeping this Issue open, because I need to make FreeTouchDeck compatible with the newest version of the Arduino Core.

Dark1886, 2024-08-06:
@DustinWatts This issue specifically mentioned an MQTT broker, but would it be possible to have the FreeTouchDeck just be an MQTT client that could execute some commands if it receives an MQTT message from an external broker?

closed 2024-10-13 by github-actions[bot]